QuinnDamerell/OctoPrint-OctoEverywhere #119: Elegoo Centauri Carbon and Spoolman

Biochao, 2025-07-13:
Does the Spoolman integration support the Centauri Carbon? 
I have OctoEverywhere and Spoolman setup with Docker in Unraid and I am trying to get the Spoolman integration set up following this setup process (https://octoeverywhere.com/spoolman) it says I 'need at least one printer set up'. I have the Centauri Carbon set up with OctoEverywhere and it works great and during a print I can see the filament used in the quick view. Going to the printer setup on the next page and selecting 'Spoolman Filament Manager' just results in a loop going back to the first step of the Spoolman integration setup process.

Johnnycyan, 2025-07-31:
I would just like to report that I also have the same problem. And I realize that octo would have no way of knowing which filament I was using but like it could just ask me during printing, it could even send a notification saying "Please select which filament you are using for this print" when you start a print with spoolman connected. Theoretically, and as a programmer, I see no reason this couldn't work cause it's just API calls.

QuinnDamerell, 2025-07-31:
Thanks for posting! I think I just fixed this, so try it again and let me know if it works for you!